dancol90/ESP8266Ping #2: DEBUG_PING

carlosrodriguesIT, 2016-08-27:
Hello,

How can i see the values of the DEBUG_PING(), like tresp_time and ttotal_time?

Thanks in advance

dancol90, 2016-08-27:
Hi!

Update to the last commit and define ENABLE_DEBUG_PING before including the library in your sketch, like this:

```C++
#define ENABLE_DEBUG_PING
#include <ESP8266Ping.h>
```

carlosrodriguesIT, 2016-08-27:
OK, but how to call in the main file? And display the values?

dancol90, 2016-08-27:
It's a debug statement, so it will be called during the execution of `Ping.ping()`. When enabled, the informations will be printed on Serial.
 
At the moment, there's no other way to access `ping_resp` structure from outside the library. The only information available is the average response time, using `Ping.averageTime()`.

carlosrodriguesIT, 2016-08-27:
Yes I see, maybe construct other class like `PingValuesClass PingValues;` to show the other values, what do you think?

dancol90, 2016-08-27:
I will think about implementing a user-defined callback that gets called on every response, that will pass a reference to the `ping_resp` structure.

carlosrodriguesIT, 2016-08-27:
Yes , the people would love , would be much more complete!

I will wait for this implementation, thanks a lot for your time (Y)

dancol90, 2016-10-13:
I've realised that the define method above does not work, I'm sorry!
In #4 I've suggested a workaround to make the debug prints works.

dancol90, 2016-12-22:
Debug info print issue fixed by c5f0be4

closed 2016-12-22 by dancol90