dancol90/ESP8266Ping #4: How to use DEBUG_PING ?

Phonevilai, 2016-10-13:
i don't know use DEBUG_PING

dancol90, 2016-10-13:
There's no need to directly use DEBUG_PING, it's an internal define.
If you want to see some debug information through the serial terminal, just add this at the top of the sketch (before the `#include`s):
```c++
#define ENABLE_DEBUG_PING
```

dancol90, 2016-10-13:
Ok I've found a bug, if you do as I said in the previous message it will not work.
For now, you'll have to add the define above directly in ESP8266Ping.h file.
I'll think about a solution.

micooke, 2016-11-15:
Hey, you need to move all uses of DEBUG_PING into the header. Easiest way is to make it a header only library.

Arduino compile and link is 'special', in that it compiles the library separately and links it in. This means that the cpp only knows about the headers it has included and not the base ino that is using it in the first place.

Great work btw :thumbsup:

dancol90, 2016-11-15:
Hi, thanks!
Yes, I know that.
I haven't had any time yet to fix this problem...I will look into it as soon as I can!

dancol90, 2016-12-22:
Commit c5f0be4 fixes the issue.

closed 2016-12-22 by dancol90