dancol90/ESP8266Ping #5: Ping fails

Jimbobnz, 2016-10-13:
Every ip address and host names are failing to ping.
the Ping.averageTime() returns zero.

dancol90, 2016-10-13:
Hi!

Are you using Arduino IDE or PlatformIO? Which version?
Have you updated the ESP8266 package to the latest version?

Jimbobnz, 2016-10-13:
I'm using Arduino 1.6.11 on Windows 10.
ESP8266 Arduino core version 2.2.0.

dancol90, 2016-10-13:
I've tried the HostPing example with IDE 1.6.12 and core 2.2.0 and it works.
Maybe is a network issue. Can the ESP8266 reach the internet or any local server (with HTTP request, for example)?

Jimbobnz, 2016-10-13:
Okay, I sort of isolated the problem with no real solution. The issue a combination of my ESP-12E and my Wireless Access Point. It's able to connect okay and get a IP address but it is unable to do any TCP or UDP communication. 
I am able to ping the ESP-12E from my PC but that about it. However if I change Access Points on the same LAN it works. Bizarre.

Since this is not a problem related to your code you can close this case.
Thanks.

closed 2016-10-14 by dancol90