dancol90/ESP8266Ping #8: Suggestion: retrieve ip address if ping to host was successful

vossilius, 2017-01-03:
Thanks for the library! Just a thought. Would be nice if one could retrieve the ip address from a ping reply to a host in order to save another dns request. I use the function to select one out of three servers based on their latency and whould like to use the returned ip address straight away instead of accessing the selected server by its host name once again.

dancol90, 2017-01-03:
Hi!

To resolve the ip address the library use the `WiFi.hostByName()` function, like this:

```Arduino
bool PingClass::ping(const char* host, byte count) {
    IPAddress remote_addr;

    if (WiFi.hostByName(host, remote_addr))
        return ping(remote_addr, count);

    return false;
}
```

I have to think about a way to return some data in a prettier way, including host IP, average time, success rate, etc.

In the meantime you can mimic the function above in your code, saving the return value of `WiFi.hostByName()`!

vossilius, 2017-01-03:
Thanks for your prompt reply. Yes, that's what I am doing right now. But doesn't that call initiate another dns request?

dancol90, 2017-01-03:
`Ping.ping()` also accept an `IPAddress` instead of a host string.

So, if you call `hostByName` in your code, and then pass the resulting IPAddress to `Ping.ping()`, one one DNS request will be made.

There's also an example (https://github.com/dancol90/ESP8266Ping/blob/master/examples/SimplePing/SimplePing.ino) that uses directly an IPAddress.

vossilius, 2017-01-03:
Ah! Got it. Thanks!

closed 2017-10-12 by dancol90