dancol90/ESP8266Ping #10: where is download esp8266wifi.h ?

njlinjie, 2017-02-10:
where is download esp8266wifi.h ? thanks!

dancol90, 2017-02-10:
Hi!
Your question is not clear, but if you want to download the library, use this link:
https://github.com/dancol90/ESP8266Ping/archive/master.zip

njlinjie, 2017-02-10:
Thank You,In your code ESP8266Ping.h

extern "C" {
  #include <ping.h>
}

dancol90, 2017-02-10:
That header file is part of the espresso SDK and it is included in the ESP8266 Arduino core (https://github.com/esp8266/Arduino).
There's no need to download it elsewhere.
If your compiler doesn't find it, you're using this library wrong or your Arduino installation is not working as it should.

njlinjie, 2017-02-10:
Thank you ,I see , My compiler is arduino-1.8.1-windows

closed 2017-10-12 by dancol90

RASHID-AMIN, 2018-12-25:
i want esp8266wifi.h library