dancol90/ESP8266Ping #12: Non-blocking version?

WayneManion, 2017-04-11:
I am using this library in a home automation project. Basically, the device pings my (local) MQTT broker and 8.8.8.8 every minute to make sure the broker and the internet connection are working. If several consecutive attempts fail, the device reboots in AP mode and waits for someone to fix it, probably by giving updated Wi-Fi credentials or a new MQTT broker address.  

It works pretty well. The problem is that the ping process is slow and blocks the device from doing anything else. I don't know enough C to look at this code and figure out exactly what is happening. Is there a way to make a version of this library where one could specify a function or instruction to run in a loop while waiting for the ping to execute? Mostly I want the device to be able to respond to MQTT messages and HTTP requests while executing a ping. 

I guess I could drop the ping frequency to once every five minutes or something, but I think there is a more elegant solution.

dancol90, 2017-04-12:
Hi,

Yes, it can be surely done. I don't have much free time these days but I'll try to take look into it!

Jesperdp, 2017-05-24:
To WayneManion

Have you tried to specify ping times?
eg:
Ping.ping(*IP or host*,1)

where the 1 is times to ping.

in my application a ping takes 4 seconds if the number of pings is unspecified.
but with the ping specified as 1 i takes 1-2 miliseconds.

beicnet, 2017-11-15:
@Jesperdp You saved me a lot of time! Thx ;)

dancol90, 2018-06-04:
Yes, as @Jesperdp said, specifing a count of 1 will avoid the wait time between two ping requests. By default this parameter is 5 and the coarse time (the interval between requests) is hardcoded to 1 second, so the call will take around 4 seconds.

closed 2018-06-04 by dancol90