dancol90/ESP8266Ping #13: WeMos D1 mini

Azuzula, 2017-04-21:
Hello, Im trying your library but it works weird at my wemos d1.
If internet run normaly, then:
Ping.ping("www.google.com") return 1
Ping.averageTime() return time about 17-20ms
But, If I disconnect cable from my ISP out of wifi router then Ping.ping still return 1 and time is 0-10ms.
And when wifi router is switched off, then Ping.ping return 0.

Here is part of my code and serial monitor.
[image]
Im using Wifi router Tenda F300.

dancol90, 2017-04-22:
Hi, that's strange!

Try adding a ```#define ENABLE_DEBUG_PING ``` a the top of the sketch and paste here what gets printed on the serial monitor.

Azuzula, 2017-04-22:
Thank you for answer but I don't know how to print debug. Im sorry for stupid question, but may I write something more into my code? Something like Serial.println(debug);?

dancol90, 2017-04-24:
Just add ```#define ENABLE_DEBUG_PING``` before the line ```#include <ESP8266Ping.h>``` at the top of your sketch, upload it as usual and then look at the Serial Monitor, just like you did for the screenshot in your message.

Azuzula, 2017-04-24:
Oh, I put it after all #define lines. My bad. This is what wemos spits on me right now.

**Cable disconected, Internet don't work:**
Pinging host
DEBUG: ping reply
	total_count = 0 
	resp_time = 1 
	seqno = 31 
	timeout_count = 0 
	bytes = 32 
	total_bytes = 0 
	total_time = 0 
	ping_err = 0 
DEBUG: ping reply
	total_count = 0 
	resp_time = 1 
	seqno = 32 
	timeout_count = 0 
	bytes = 32 
	total_bytes = 0 
	total_time = 0 
	ping_err = 0 
DEBUG: ping reply
	total_count = 0 
	resp_time = 1 
	seqno = 33 
	timeout_count = 0 
	bytes = 32 
	total_bytes = 0 
	total_time = 0 
	ping_err = 0 
DEBUG: ping reply
	total_count = 0 
	resp_time = 1 
	seqno = 34 
	timeout_count = 0 
	bytes = 32 
	total_bytes = 0 
	total_time = 0 
	ping_err = 0 
DEBUG: ping reply
	total_count = 0 
	resp_time = 1 
	seqno = 35 
	timeout_count = 0 
	bytes = 32 
	total_bytes = 0 
	total_time = 0 
	ping_err = 0 
Avg resp time 1 ms
bool ret = 1
Success ping to www.google.com - 1ms

**Cable connected, internet work:**

Pinging host
DEBUG: ping reply
	total_count = 0 
	resp_time = 17 
	seqno = 36 
	timeout_count = 0 
	bytes = 32 
	total_bytes = 0 
	total_time = 0 
	ping_err = 0 
DEBUG: ping reply
	total_count = 0 
	resp_time = 17 
	seqno = 37 
	timeout_count = 0 
	bytes = 32 
	total_bytes = 0 
	total_time = 0 
	ping_err = 0 
DEBUG: ping reply
	total_count = 0 
	resp_time = 20 
	seqno = 38 
	timeout_count = 0 
	bytes = 32 
	total_bytes = 0 
	total_time = 0 
	ping_err = 0 
DEBUG: ping reply
	total_count = 0 
	resp_time = 17 
	seqno = 39 
	timeout_count = 0 
	bytes = 32 
	total_bytes = 0 
	total_time = 0 
	ping_err = 0 
DEBUG: ping reply
	total_count = 0 
	resp_time = 17 
	seqno = 40 
	timeout_count = 0 
	bytes = 32 
	total_bytes = 0 
	total_time = 0 
	ping_err = 0 
Avg resp time 17 ms
bool ret = 1
Success ping to www.google.com - 17ms

dancol90, 2017-05-06:
That's strange, are you using the last esp8266 package? Maybe something changed in the official SDK (that this library uses).

Azuzula, 2017-05-06:
Im using IDE 1.8.1. built in library ESP8266Wifi ver. 1.0.0.
Im thinking about check if wemos can get time from NTP server instead of ping if we can't get this work.

tekhedd, 2017-05-07:
I can not reproduce the problem with the current trunk code.

  * generic clone wemos D1 mini board (not pro)
  * IDE 1.8.2
  * Netgear 7000
  * CenturyLink (C1000A) in bridge mode
  * Using 2.4G on the guest network
  * ESP8266Wifi contrib 1.0.0

I know that (at least in the previous firmware version) the netgear router would return its own IP address when DNS lookups failed, presumably as a magic autoconf feature. This resulted in successful pings  even with the modem unplugged.  Could this be the issue?

I have in fact just made my own "WIFI ResetPlug" type device, because the OTS reset plug device was fooled by this behavior and never reset my link. :) You have to do both IP and DNS-based pings to test connectivity: sometimes DNS lookups fail, sometimes the router can trick you into pinging the wrong IP. I'll be posting the code here shortly. 

In any case, can not reproduce.

Azuzula, 2017-05-07:
Im not sure why this happen, but I rather use NTP instead of ping now. Anyway Im using NTP for syncing my arduino clock connected via my NRF net.
@tekhedd My Tenda F300 shows his own setup page in browser if internet dont work. Perhaps it is main cause of my problem.

dancol90, 2017-05-07:
Yeah, it most surely is the cause of the problem. Try pinging a remote IP instead of a host name!

tekhedd, 2017-05-08:
FYI: https://github.com/tekhedd/ResetFlake  It's pretty minimal but works, has friendly console output, and pings by both DNS and IP. Might be useful for a quick test to see if this is the issue, well, for other people who find this issue by searching.  Uses ESP8266Ping, of course. :)

(I'm running this code for real at my house now, so everybody now knows they can DOS my internet uplink simply by turning off centurylink's primary DNS resolvers for the western US.)

closed 2017-10-12 by dancol90