dancol90/ESP8266Ping #16: unable to use "Ping.ping" method, doesn't compile.

kantorv, 2017-12-29:
any suggestions? the output of compilation error:

```sketch/SimplePing.ino.cpp.o: In function `PingClass::_ping_recv_cb(void*, void*)':```
```/home/user/Arduino/libraries/ESP8266Ping-master/src/ESP8266Ping.impl.h:65: undefined reference to `ping_start'```
```sketch/SimplePing.ino.cpp.o: In function `PingClass::ping(IPAddress, unsigned char)':```
```/home/user/Arduino/libraries/ESP8266Ping-master/src/ESP8266Ping.impl.h:65: undefined reference to `ping_start'```
```collect2: error: ld returned 1 exit status```
```exit status 1```
```Error compiling for board Generic ESP8266 Module.```

AndreyShamis, 2017-12-29:
Check that you dont have #include <ping.h> in your skatch
Remove it or wrap it this way
`
extern "C" {
  #include <ping.h>
}
`

kantorv, 2017-12-31:
@AndreyShamis I used `SimplePing.ino` sketch from examples folder, without modifications.

AndreyShamis, 2018-01-04:
@kantorv Can you provide full log?

dancol90, 2018-01-04:
Hi!

Have you updated the library to the latest version?
Which esp8266 arduino core are you using?

kantorv, 2018-01-07:
@dancol90 , @AndreyShamis  the issue resolved after updating `esp8266 arduino core` .. 
Thank you.

closed 2018-01-07 by kantorv