dancol90/ESP8266Ping #19: Facing issues when ping count > 250 in SimplePing.ino

2BECKHAM2, 2018-05-16:
Thanks for the nice library dancol90.

Upon checking I observed that the number of ping requests are not initiated based on the ping count that I configured.

********* CODE ***
#include <ESP8266WiFi.h>
#include <ESP8266Ping.h>
#include <Streaming.h>

  
  const int ping_count = 1100;
  const IPAddress remote_ip(8, 8, 8, 8);

void setup() {
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println("Connecting to WiFi");
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.print(".");
  }

  Serial.println();
  Serial.print("WiFi connected with ip ");  
  Serial.println(WiFi.localIP());

   Serial << "Ping count configured : " << ping_count << "," << " Pinging ip : " << remote_ip <<endl;
 

   if(Ping.ping(remote_ip, ping_count)) {
    Serial.println("Success!!");
  } else {
    Serial.println("Error :(");
  }
}

void loop() { }

****************************************

When configured the ping count = 300, the ping requests are initiated only up to 144 times. The expected behavior is 300 times. I've listed the log details below.

*****************
WiFi connected with ip 192.168.31.178
Ping count configured : 300, Pinging ip : 8.8.8.8
Reply from:  total_count = 0  resp_time = 48ms  seqno = 1  timeout_count = 0  bytes = 32  total_bytes = 0  total_time = 0  ping_err = 0 
.....

Reply from:  total_count = 0  resp_time = 48ms  seqno = 44  timeout_count = 0  bytes = 32  total_bytes = 0  total_time = 0  ping_err = 0 
Avg resp time 49 ms
Success!!

*******************
When configured for 400

WiFi connected with ip 192.168.31.178
Ping count configured : 400, Pinging ip : 8.8.8.8
Reply from:  total_count = 0  resp_time = 49ms  seqno = 1  timeout_count = 0  bytes = 32  total_bytes = 0  total_time = 0  ping_err = 0 
......
Reply from:  total_count = 0  resp_time = 58ms  seqno = 144  timeout_count = 0  bytes = 32  total_bytes = 0  total_time = 0  ping_err = 0 
Avg resp time 55 ms
Success!!
*************************
When configured for 1000.

Reply from:  total_count = 0  resp_time = 55ms  seqno = 231  timeout_count = 0  bytes = 32  total_bytes = 0  total_time = 0  ping_err = 0 
Reply from:  total_count = 0  resp_time = 50ms  seqno = 232  timeout_count = 0  bytes = 32  total_bytes = 0  total_time = 0  ping_err = 0 
Avg resp time 114 ms
Success!!

*************************
Note that I am facing this issue only when I set the ping count to be > 250.

dancol90, 2018-06-04:
Hi!
Yes, this is expected, as the count parameter is of type byte. Hence it can hold at most 255.
Checking the SDK headers, though, the ping_option struct uses uint32s, so maybe changing the ping function signature will suffice. I will give this a try as soon as possibile.

2BECKHAM2, 2018-06-04:
Many thanks dancol90 for the update.

commit 564a3147 (2019-12-26): Fix #19
commit 857b9488 (2023-10-27): Fix #19
closed 2019-12-26 by dancol90