dancol90/ESP8266Ping #25: Average Time, Debug Infos ?

Greece2, 2020-01-13:
How can one enable the debug function 
of this lib
`ENABLE_DEBUG_PING` 
without having to make changes to the lib itself ?
How can one print aditional info like number of errors, average time etc. ?
Nothing about that in the Demos.

Thanks !  I'm a newby, so excuse my questions ;-)

dancol90, 2020-01-13:
Hi! Just add the define before including the library in your sketch:

```c++
#define ENABLE_DEBUG_PING
#include <ESP8266Ping.h>
```

Greece2, 2020-01-14:
Hi!  I'll give that a try, thanks !
I found out that
`    int avg_time_ms = Ping.averageTime();
    Serial.print("Av.Time: ");
    Serial.println(avg_time_ms);`
prints the average repsonse time :-)

closed 2020-07-31 by dancol90