dancol90/ESP8266Ping #31: Please make ping async

sblantipodi, 2020-07-07:
As title,
pinging requires a lot of time with this lib.

Is it possible to make it async?

skorokithakis, 2020-08-10:
Seconded, I want to do some work while the library is pinging and right now it's not possible.

maiksicks, 2020-11-24:
@sblantipodi You could call `Ping.ping(host, 1);` to only send one ping request. This works quite fast.

sblantipodi, 2020-11-24:
> @sblantipodi You could call `Ping.ping(host, 1);` to only send one ping request. This works quite fast.

I'm running a stepper motor, this cause huge rattle

ziqbal, 2021-02-23:
@skorokithakis @sblantipodi
You all probably figured it out but I had this same issue past week and resolved it by using another library: [AsyncPing](https://github.com/akaJes/AsyncPing
)
It works a treat.

skorokithakis, 2021-02-23:
@ziqbal  Hey Zafar! Long time! Thanks for the tip, I'll use that.

sblantipodi, 2021-02-23:
@maiksicks it's fast but if you are driving a stepper motor on a single core microcontroller it creates a rattle every time you do a ping. :) Not an enormous issues but an async call could be cool.

closed 2023-10-27 by dancol90

sblantipodi, 2023-10-28:
@dancol90 wow you close this issue, can I ask you how to "make it async"?
should I change somethig in my code?

is the 
`bool ret = Ping.ping("www.google.com");`
now async?

maiksicks, 2023-10-29:
@sblantipodi What about this? :)
https://github.com/dancol90/ESP8266Ping/pull/47