dancol90/ESP8266Ping #39: ping library for ESP32?

johnerrington, 2021-12-27:
I've been using this library on a NODE (ESP8266) but I now need to move the project to an ESP32. Would it be possible to extend the library to work on the ESP32?

bwjohns4, 2021-12-28:
Take a look at this one https://github.com/marian-craciunescu/ESP32Ping it was forked from this library and works almost the same.

johnerrington, 2021-12-28:
Thanks for your prompt reply.  I found it and now have it working; and I've even been able to extend it to provide ping stats - success, fails, average ping time.
However I'd like to over-ride the defaults in the library (your ping.cpp)- particularly the delay  between pings - which will lock up my code
  if(ping_seq_num < count){
            delay( interval*1000L);
which I'd propose to change to delay(interval) and pass the "interval" in milliseconds.

and also reduce the timeout, which ;  but I cant see how to do that
// Setup socket
    struct timeval tout;

    // Timeout
    tout.tv_sec = timeout;
    tout.tv_usec = 0;

    if (setsockopt(s, SOL_SOCKET, SO_RCVTIMEO, &tout, sizeof(tout)) < 0) {
        closesocket(s);
        // TODO: error
        return false;
    }

and I would need to overload the ping.ping function to allow passing those parameters; and I confess my oop is not  up to working that out. 

I'll bwe very grateful for any help you can offer - and perhaps Marian could be involved?

dancol90, 2023-10-27:
Please move this issue to the ESP32Ping project, as it is unrelated to this repository.

closed 2023-10-27 by dancol90