dancol90/ESP8266Ping #48: Disabling send ping data on the serial port

saman-taghipour, 2023-11-10:
Hi,
I don't want information to be sent on the serial port
​How can I disable it?

dancol90, 2023-11-10:
The library will print to Serial only if manually enabled. Make sure you don't have any `#define ENABLE_DEBUG_PING` in your sketch

saman-taghipour, 2023-11-10:
> The library will print to Serial only if manually enabled. Make sure you don't have any `#define ENABLE_DEBUG_PING` in your sketch

I did'nt enable in my sketch, but it was enabled in ESP8266Ping.h file, in line 31
`#ifdef ENABLE_DEBUG_PING
  #define DEBUG_PING(...) Serial.printf(__VA_ARGS__)
#else
  #define DEBUG_PING(...)
#endif`

when I comment this line, sketch error occures!
`Error compiling for board Generic ESP8266 Module.
`

imayoda, 2024-02-26:
> The library will print to Serial only if manually enabled. Make sure you don't have any `#define ENABLE_DEBUG_PING` in your sketch
+1
have the same behaviour here.. it gives me some debug infos about last ping activity, even if no "debug on" was specified..
Tried event to comment out all code about it (ifdef's and cpp relatives), it compiles fine but spitts out same debug output on serial.

Got to thing is related to ping.h activity from sdk ???
Somebody got some clues?
Thanks :)

lewanox, 2024-05-23:
I have the same problem. I even tried to delete every DEBUG_PING(...), os_printf(...), LWIP_DEBUGF(...) lines, and also the ENABLE_DEBUG_PING, PING_DEBUG, etc definitions from ESP8266Ping.h, ESP8266Ping.cpp, ping.h, esp-ping.c files.
Also tried undefining the ENABLE_DEBUG_PING line.

Serial monitor shows the same type of debug messages:
> ping 1, timeout 1, total payload 0 bytes, 1015 ms

I also tried deleting Arduino IDE core and library cache, but that didn't solve the problem either.

imayoda, 2024-05-23:
> I have the same problem. I even tried to delete every DEBUG_PING(...), os_printf(...), LWIP_DEBUGF(...) lines, and also the ENABLE_DEBUG_PING, PING_DEBUG, etc definitions from ESP8266Ping.h, [...]

It's a bug somewhere with or in esp8266 core, as using 3.0.x "solves" the problem and debug output is silent.. 
Somebody has to open an issue but I'm not quite sure where to... :blush:

lewanox, 2024-05-23:
> It's a bug somewhere with or in esp8266 core, as using 3.0.x "solves" the problem and debug output is silent.. Somebody has to open an issue but I'm not quite sure where to... 😊

But even if it is a bug, I tried to find and delete every line of code that is outputting anything similar to the "ping x, timeout x, total payload x bytes, x ms" pattern, and it didn't help. I even looked through the ESP8266 SDK with a file content searcher software to find the code line but I didn't find anything yet. It can not magically spawning in there, right? 😂

imayoda, 2024-05-23:
maybe the code author @dancol90  knows best if debug relies on some debug variable from the core... :)
Can't belive such a thing doesn't impact on a larger user base.. noisy hw serial easily bloats whatever project that needs 2 mcu talking each other.